jbuckmccready/cavalier_contours #10: Improve intersect function documentation

jbuckmccready, 2021-03-07:
The functions in `cavalier_contours/circle_circle_intersect.rs`, `cavalier_contours/line_circle_intersect.rs`, `cavalier_contours/line_line_intersect.rs`, and `cavalier_contours/pline_seg_intersect.rs` need code examples added to their documentation. E.g. `circle_circle_intr` just has `/// Finds the intersects between two circles.`

Add more documentation about their parameters, and add a basic code example similar to what is done for the functions in the `cavalier_contours/pline_seg.rs` module.

Nevsden, 2022-01-04:
Also wanted to mention, that the way you have written the macro `assert_case_eq` (e.g. for `line_line_intr`) in your testing suite is neat!

jbuckmccready, 2022-08-22:
Improved with PR https://github.com/jbuckmccready/cavalier_contours/pull/18, thanks @Nevsden.

closed 2022-08-22 by jbuckmccready