jbuckmccready/cavalier_contours #8: Add characteristic polylines and more benchmarks

jbuckmccready, 2021-03-07:
More benchmarks should be added for the core polyline functions:
* Polyline path length
* Polyline parallel offset
* Polyline winding number
These benchmarks can be added to `cavalier_contours/benches/benches.rs`.

More polylines (capturing different characteristics, e.g. polyline with lots of lines, polyline with lots of arcs, polyline when offset that has many self intersects, polyline when offset that has very little self intersects, etc.) should be added and available for the benchmarks as well. These can be added to `cavalier_contours/benches/test_polylines.rs`.

As things grow it may make sense to create another crate in the workspace that has all the characteristic polylines we want to add for testing so it may be accessed from the other crates (test_suite, etc.).

denisraison, 2021-03-07:
Hey @jbuckmccready,

It's me back again :D.

I'd like to contribute to your work as much as I can.
As I've never used rust before would you mind giving me some insights on how you're testing and running the library?

Thanks

jbuckmccready, 2021-03-07:
@denisraison Hello! For getting started I recommend the Rust book getting started [here](https://doc.rust-lang.org/book/ch01-00-getting-started.html) and reading through the chapters. That will get you up and going with cargo to build the library and run tests.

I use Visual Studio Code with the `rust-analyzer` extension for editing.

This library follows the all the standard cargo conventions so `cargo test` will run all the tests, `cargo bench` will run all the benchmarks, etc.

jbuckmccready, 2021-03-07:
The benchmarks use the criterion library which you can read about [here](https://github.com/bheisler/criterion.rs) and you can also just look at the existing code and add to the structure. You can also see some example use of criterion benchmark code [here](https://github.com/jbuckmccready/static_aabb2d_index/blob/master/benches/bench_static_aabb2d_index.rs) which is from the spatial index crate this library uses.

jbuckmccready, 2021-03-15:
I actually decided to move the benchmarks out of this repository since criterion slowed the fresh dev build significantly. I am closing this issue for now (I plan to make another git repository that references this one solely for benchmarks sometime in the future). If you'd like to contribute please check out the other issues.

closed 2021-03-15 by jbuckmccready