jbuckmccready/cavalier_contours #11: Can you build a wasm library to use

FishOrBear, 2021-06-16:


jbuckmccready, 2021-06-16:
I don't have any plans currently to work on and maintain a wasm API but Rust supports wasm directly, see the web demo project [here](https://github.com/jbuckmccready/cavalier_contours_web_demo). The example file that creates an FFI for wasm is [here](https://github.com/jbuckmccready/cavalier_contours_web_demo/blob/master/wasm/cavalier_contours_web_ffi/src/lib.rs).

FishOrBear, 2021-06-17:
The algorithm ran in my WebCAD project, and the performance was very good.

https://user-images.githubusercontent.com/19372111/122324043-b1af6400-cf5a-11eb-80c1-cd8dfc21dee4.mp4

FishOrBear, 2021-06-17:
vers: [0, 0, 0, -600, 0, 0, -600, -18, 0, -18, -18, 0, -18, -600.0000000000005, 0, 0, -600.0000000000005, 0]
close:false

There is a problem with this result

jbuckmccready, 2021-06-17:
Yeah the wasm performance is quite good, the the interactive web demo app for cavalier contours is [here](https://jbuckmccready.github.io/cavalier_contours_web_demo_page/).

Can you give me the minimum reproducible data for the problem you found (vertexes, is_closed, offset value)?

jbuckmccready, 2021-06-17:
I figured it out from your post + video, I'll look into it. Here it is in the cavalier contours web demo app:
```json
{
  "isClosed": false, 
  "vertexes": [
    [0, 0, 0],
    [-600, 0, 0],
    [-600, -18, 0],
    [-18, -18, 0],
    [-18, -600.0000000000005, 0],
    [0, -600.0000000000005, 0]
  ]
}
```
[image]

FishOrBear, 2021-06-17:
[image]

This fillet is sometimes not needed.
AutoCAD's offset does not handle this.

Can you add a control field to control whether or not to chamfer?

jbuckmccready, 2021-06-17:
Possibly, I don't currently have a use for that so I'm not sure when I'll get around to it. The algorithm performs distance checks between the offset segments and the original polyline to to prune collapsed regions, this may get falsely triggered in some cases if the lines are extended out rather than rounded (if the lines are truncated with a straight edge this wouldn't falsely trigger). I'd have to look into it more.

Is there a hosted interactive page for the CAD tool you're creating?

FishOrBear, 2021-06-17:
https://t.cfcad.cn/offset_wasm/ This branch uses the wasm library.
https://t.cfcad.cn/ This branch is offset by js.

You need an account
Account: jbuckmccready
Password: cready2021

jbuckmccready, 2021-06-17:
Page never finishes loading - download rate seems to be getting throttled/cut, I just see the loading screen and animation but it's stuck at 99%.

FishOrBear, 2021-06-17:
Our server is in China, and foreign access may be slower.

What chat software do you use, or can you use teamview to remotely assist?

Or open f12 to see if the js resource can be loaded normally

jbuckmccready, 2021-06-17:
I'm in US California, I think it's just blocking traffic from where I am, when I look at task manager network activity is always less than 0.1Mbps. I tried both Firefox and Microsoft Edge.

jbuckmccready, 2021-06-17:
I looked into the issue you reported, I don't see any problems. The polyline you posted seems to offset correctly when either open or closed. What offset value are you using? In the video it looks like it's collapsed to a coincident segment (offset = 9) and everything looks correct in the cavalier contours web demo (the algorithm always keeps coincident/overlapping results).

FishOrBear, 2021-06-17:
Try to modify the hosts

47.106.224.252 t.cfcad.cn

FishOrBear, 2021-06-17:
{"vertexes":[[0,0,0],[-600,0,0],[-600,-18,0],[-18,-18,0],[-18,-600.0000000000005,0],[0,-600.0000000000005,0],[0,0,0]],"isClosed":false}

offset = 9

FishOrBear, 2021-06-17:
When isClosed=true, the result is correct.

FishOrBear, 2021-06-17:
So `isClose` does not seem to be a closed mark `closeMark`

The behavior is different from autocad.

jbuckmccready, 2021-06-17:
Even when open it produces the correct result when I tested it. The vertical edge just lines up with the origin axis line so it looks wrong visually in the screenshot, but it's an open polyline so the part sticking out past the arc should be there.

How do you expect the the result should be different?

FishOrBear, 2021-06-17:
I thought their behavior should be consistent.

Because the contour is the same when isClose=true or IsClose=false.

So their offset result should be the same.

FishOrBear, 2021-06-17:
{"vertexes":[[0,0,0],[100,0,0],[100,100,0],[0,100,0]],"isClosed":true}

{"vertexes":[[0,0,0],[100,0,0],[100,100,0],[0,100,0],[0,0,0]],"isClosed":false}

I think their offset results should be the same, but they are not the same.

FishOrBear, 2021-06-17:
But this is not a problem, I can adapt this by changing the code

jbuckmccready, 2021-06-17:
Pasting those examples into the cavalier contours web demo produces the same result... Am I missing something?
[image]
[image]

jbuckmccready, 2021-06-17:
Oh are you saying the "isClosed" property isn't set to true in the result? Yes if the input is an open polyline then it returns it as an open polyline. If you want to close it you can test if `polyline.firstVertexPosition == polyline.lastVertexPosition`.

FishOrBear, 2021-06-17:
Oh, it looks right, I did it wrong, I'm very sorry

FishOrBear, 2021-06-17:
{"vertexes":[[11297.108605108433,3514.1617009614806,0],[11411.528713321535,3514.1617009614806,0],[11411.528713321535,3647.1628523886743,0],[11297.108605108433,3647.1628523886743,0],[11297.108605108433,3514.1617009614806,0]],"isClosed":true}

[image]

This doesn't look great

FishOrBear, 2021-06-17:
> Pasting those examples into the cavalier contours web demo produces the same result... Am I missing something?
> [image]
> [image]

What is this url address

FishOrBear, 2021-06-17:
{"vertexes":[[0,0,0],[100,0,0],[100,100,0],[0,100,0],[0,0,0]],"isClosed":true}

[image]

This doesn't look great

FishOrBear, 2021-06-17:
[image]
```
{"vertexes":[[12423.122820176195,3528.5133862542775,0],[12423.122820176195,3510.5133862542775,0],[12837.342864924894,3510.5133862542775,0],[12837.342864924894,2907.7942631500323,0],[12855.342864924894,2907.7942631500323,0],[12855.342864924894,3528.5133862542775,0]],"isClosed":true}
```

It should be better to return one line in this case

jbuckmccready, 2021-06-17:
https://jbuckmccready.github.io/cavalier_contours_web_demo_page/ for the web demo page.

jbuckmccready, 2021-06-17:
> 
> 
> {"vertexes":[[0,0,0],[100,0,0],[100,100,0],[0,100,0],[0,0,0]],"isClosed":true}
> 
> [image]
> 
> This doesn't look great

The offset algorithm does not test for/handle repeat position vertexes (here x=0, y=0 is repeated). You can use the `remove_repeat_pos` polyline method to clean it up before performing the offset.

I plan to add a offset options flag that will cause the function to clean things up as best it can before performing the offset, currently it doesn't do it for performance reasons. Note it does have a debug assert to catch this case, I recommend compiling with debug asserts enabled for testing.

jbuckmccready, 2021-06-17:
> [image]
>
> ``` [...]

The algorithm tries to join the longest connected sequence when overlapping is involved. Can you create a simpler example that is smaller and has the problem you describe?

FishOrBear, 2021-06-17:
```

{"vertexes":[[0,0,0],[432.22004474869937,0,0],[432.22004474869937,-620.7191231042452,0],[414.22004474869937,-620.7191231042452,0],[414.22004474869937,-18,0],[0,-18,0]],"isClosed":true}

```

It looks like the results overlap

https://user-images.githubusercontent.com/19372111/122344250-2c877780-cf79-11eb-849d-c1f05fd96f4d.mp4

jbuckmccready, 2021-06-17:
I think I found the issue overlap join issue and I have a fix working. I will try to review and commit it tomorrow, and add the case to the automated tests.

FishOrBear, 2021-06-17:
include/cavc/polylineoffsetislands.hpp

It seems that this feature has not been implemented in rust, and I haven't found this method.

FishOrBear, 2021-06-18:
https://user-images.githubusercontent.com/19372111/122493643-a4f24500-d01a-11eb-97f8-f2e26873c0cd.mp4

This polyline cannot be offset outside

```

{"vertexes":[[0,0,2.9239855509708432],[422.7001801508977,-2181.678349165927,0.21373279590439873],[1217.850643134374,-1733.2038795014323,4.472986753391631],[961.0754444813574,-155.87051634718955,0.22947158823257166]],"isClosed":true}

```

jbuckmccready, 2021-06-18:
I committed fix and added test case for issue you had with overlapping result, test it and let know how it goes.

I'm closing this issue, please make a new issues for each bug report, it's easier for me to track things

closed 2021-06-18 by jbuckmccready