jbuckmccready/cavalier_contours #16: Overlapping lines appear in the offset result

FishOrBear, 2021-06-24:
```
{"vertexes":[[0,0,0],[0,-882.3073569599583,0.19403531582953024],[30.99567883971031,-942.6919889602286,0.1940353158226031],[0,-1003.0766209573267,0],[0,-1880,0],[4,-1880,0],[4,-1003.2606547873729,-0.19502429444774674],[34.00003211540752,-945.3840053746535,-0.19502429171847763],[64,-1003.2606547872638,0],[64,-1880,0],[68,-1880,0],[68,-1003.0766209572321,0.1940353134888888],[37.00432115854346,-942.6919889585697,0.194035313489369],[68,-882.3073569597691,0],[68,0,0],[64,0,0],[64,-882.1233231297883,-0.19502431066879408],[33.999968577962136,-939.9999731353018,-0.19502429444816094],[4,-882.1233231296792,0],[4,0,0]],"isClosed":true}
```

jbuckmccready, 2021-06-24:
It purposely keeps all overlapping results currently, I think it makes sense to add an option that allows toggling keeping coincident/overlapping results or discarding them.

jbuckmccready, 2021-06-24:
I created an issue for such an option here: https://github.com/jbuckmccready/cavalier_contours/issues/17

Does that address what you're referring to or was there something else you meant?

FishOrBear, 2021-06-24:
Ok.
there is no problem

closed 2021-06-24 by FishOrBear