jbuckmccready/cavalier_contours #19: Support polyline with multiple offsets

WindingWinter, 2022-01-06:
I [posted this question](https://github.com/jbuckmccready/CavalierContours/issues/47) at the C++ issue tracker-- I'm not sure whether this or that place is the right place to post. Anyway feel free to close the one that is not relevant.

The question is this: Is there any plan to support variable offsets? Something like [NetTopologySuite VariableBuffer](https://nettopologysuite.github.io/NetTopologySuite/api/NetTopologySuite.Operation.Buffer.VariableBuffer.html#NetTopologySuite_Operation_Buffer_VariableBuffer__ctor_NetTopologySuite_Geometries_Geometry_System_Double___). Currently only uniform offset can be done throughout all the vertexes on the polylines.

jbuckmccready, 2022-01-06:
I think the algorithm used by this library would not work with variable offset distance because distance checks are used to remove invalid/collapsed offset parts (global invalid segments), I am not sure what that means when comparing distances between two variable offset distance segments - it would definitely have to be modified to get it to work. There may be other modifications required to make the raw offset join/trim code work as well.

I currently have no plan to to work on variable offsets, but pull requests/contributions are welcome if the algorithm is implemented and working.

jbuckmccready, 2022-01-08:
I don't have any ideas offhand - I think it would require a fairly deep dive into it.

FishOrBear, 2022-04-26:
Maybe something like this?
[image]

WindingWinter, 2022-10-11:
Yes, but is that doable?