jbuckmccready/cavalier_contours #25: Failing test case from the web version

galou, 2022-11-25:
Feel free to close immediately, I didn't search if the bug was already submitted.

I just played with the web version of CavalierContours and found a failing test:
```(pline_closed![(165.5, 179.0, -0.5),
(80.0, 90.0, 0.374794619217547),
(231.5, -278.0, 0.0),
(230.0, 0.0, 1.0),
(320.0, 0.0, -0.5),
(280.0, 0.0, 0.5),
(390.0, 210.0, 0.0),
(280.0, 120.0, 0.5)], 3.0) =>
                    [PlineProperties::new(11, 0.0, 1563.689560505266, 66.99239238371496, -273.0909692941639, 401.41586988912127, 205.22199935960901)]
```
with mode *Offset* , Offset 3, Offset count 50, *Handle Self Intersects* activated.

Thanks for offering this project!

closed 2022-11-25 by galou

jbuckmccready, 2022-11-26:
You closed this issue already, I haven't looked at it, is this an actual bug?

galou, 2022-12-03:
I think that my path was then self-intersecting and I guess that the offset would have been in one of the loop if the loop were large enough.