jbuckmccready/cavalier_contours #27: Problems with debug asserts

tredecimguttatus, 2023-03-19:
Hello!
I've continued investigations from topic https://github.com/jbuckmccready/cavalier_contours/issues/23

Got several new results.

Input vectors:
```
{
  "name": "intersected_initial",
  "isClosed": true,
  "Area": 4.155,
  "vertexes": [
    [68.969210062592879, 39.825009999999999, 0.227652821420176],
    [71.447072586049913, 41.014860011263892, 1.000000000000000],
    [70.455927413950079, 41.808919988736108, -0.227652821420176],
    [68.969209937407129, 41.095009999999995, 1.000000000000000]
  ]
}

{
  "name": "result_0",
  "isClosed": true,
  "Area": 11.3863,
  "vertexes": [
    [71.447351802797868, 41.015208780804905, 0.000000000000000],
    [75.004341802797867, 45.461448780804908, 1.000000000000000],
    [74.012638197202122, 46.254811219195098, 0.000000000000000],
    [70.455648197202123, 41.808571219195095, -0.227591152597540],
    [68.969209937407129, 41.095009999999995, 1.000000000000000],
    [68.969210062592879, 39.825009999999999, 0.227652821420176],
    [71.447072586049913, 41.014860011263892, 0.000175893156428]
  ]
}
```

In debug mode i see assert failure:

```
thread '<unnamed>' panicked at 'assertion failed: pline.at(0).pos().fuzzy_eq_eps(pline.last().unwrap().pos(), slice_join_eps)', C:\cavalier_contours-master\cavalier_contours\src\polyline\internal\pline_boolean.rs:605:9
stack backtrace:
   0:     0x7ffd5310f20f - std::backtrace_rs::backtrace::dbghelp::trace
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\..\..\backtrace\src\backtrace\dbghelp.rs:98
   1:     0x7ffd5310f20f - std::backtrace_rs::backtrace::trace_unsynchronized
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\..\..\backtrace\src\backtrace\mod.rs:66
   2:     0x7ffd5310f20f - std::sys_common::backtrace::_print_fmt
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\sys_common\backtrace.rs:66
   3:     0x7ffd5310f20f - std::sys_common::backtrace::_print::impl$0::fmt
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\sys_common\backtrace.rs:45
   4:     0x7ffd53121baa - core::fmt::write
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\core\src\fmt\mod.rs:1196
   5:     0x7ffd5310dac9 - std::io::Write::write_fmt<std::sys::windows::stdio::Stderr>
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\io\mod.rs:1654
   6:     0x7ffd53110e5b - std::sys_common::backtrace::_print
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\sys_common\backtrace.rs:48
   7:     0x7ffd53110e5b - std::sys_common::backtrace::print
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\sys_common\backtrace.rs:35
   8:     0x7ffd53110e5b - std::panicking::default_hook::closure$1
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\panicking.rs:295
   9:     0x7ffd53110a4e - std::panicking::default_hook
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\panicking.rs:314
  10:     0x7ffd53111451 - std::panicking::rust_panic_with_hook
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\panicking.rs:698
  11:     0x7ffd531112d2 - std::panicking::begin_panic_handler::closure$0
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\panicking.rs:586
  12:     0x7ffd5310fb17 - std::sys_common::backtrace::__rust_end_short_backtrace<std::panicking::begin_panic_handler::closure_env$0,never$>
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\sys_common\backtrace.rs:138
  13:     0x7ffd53110fe9 - std::panicking::begin_panic_handler
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\panicking.rs:584
  14:     0x7ffd53128225 - core::panicking::panic_fmt
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\core\src\panicking.rs:142
[... 94 lines cut ...]
  22:     0x7ffd531e6c61 - static_aabb2d_index::static_aabb2d_index::StaticAABB2DIndex<f64>::visit_query_with_stack<f64,cavalier_contours::polyline::internal::pline_intersects::find_intersects::closure_env$0<cavalier_contours::polyline::pline::Polyline<f64>,cavalier_contours::polyl
                               at C:\Users\KarakurT\.cargo\registry\src\github.com-1ecc6299db9ec823\static_aabb2d_index-1.0.0\src\static_aabb2d_index.rs:1012
  23:     0x7ffd5324116e - cavalier_contours::polyline::internal::pline_intersects::find_intersects<cavalier_contours::polyline::pline::Polyline<f64>,cavalier_contours::polyline::pline::Polyline<f64>,f64>
                               at C:\cavalier_contours-master\cavalier_contours\src\polyline\internal\pline_intersects.rs:402
  24:     0x7ffd53222bdb - cavalier_contours::polyline::internal::pline_boolean::process_for_boolean<cavalier_contours::polyline::pline::Polyline<f64>,cavalier_contours::polyline::pline::Polyline<f64>,f64>
                               at C:\cavalier_contours-master\cavalier_contours\src\polyline\internal\pline_boolean.rs:55
  25:     0x7ffd53220774 - cavalier_contours::polyline::internal::pline_boolean::polyline_boolean<cavalier_contours::polyline::pline::Polyline<f64>,cavalier_contours::polyline::pline::Polyline<f64>,cavalier_contours::polyline::pline::Polyline<f64>,f64>
                               at C:\cavalier_contours-master\cavalier_contours\src\polyline\internal\pline_boolean.rs:744
  26:     0x7ffd532392ca - cavalier_contours::polyline::traits::PlineSource::boolean_opt<cavalier_contours::polyline::pline::Polyline<f64>,cavalier_contours::polyline::pline::Polyline<f64> >
                               at C:\cavalier_contours-master\cavalier_contours\src\polyline\traits.rs:1465
  27:     0x7ffd53236bbc - cavalier_contours_ffi::cavc_pline_boolean::closure$0
                               at C:\cavalier_contours-master\cavalier_contours_ffi\src\lib.rs:965
  28:     0x7ffd5327d120 - std::panicking::try::do_call<cavalier_contours_ffi::cavc_pline_boolean::closure_env$0,i32>
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3\library\std\src\panicking.rs:492
  29:     0x7ffd5327f193 - std::panicking::try::do_catch<cavalier_contours_ffi::cavc_plinelist_pop::closure_env$0,i32>
  30:     0x7ffd5327c68c - std::panicking::try<i32,cavalier_contours_ffi::cavc_pline_boolean::closure_env$0>
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3\library\std\src\panicking.rs:456
  31:     0x7ffd53211d0d - std::panic::catch_unwind<cavalier_contours_ffi::cavc_pline_boolean::closure_env$0,i32>
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3\library\std\src\panic.rs:137
  32:     0x7ffd532a1c11 - cavalier_contours_ffi::cavc_pline_boolean
                               at C:\cavalier_contours-master\cavalier_contours_ffi\src\lib.rs:949
```

Best regards,
Andrei Rodionov

jbuckmccready, 2023-03-19:
What version (or commit) of cavalier contours are you using? I was unable to trigger a debug assert from those inputs using the latest on master (commit: 08200b5b9856eb030266db0a57700830a39b3ff5).

tredecimguttatus, 2023-03-19:
Hello!

Thats interesting, i've checked asserts after using this scripts:
```
cd C:\
rd C:\cavalier_contours-master /s /q
git clone https://github.com/jbuckmccready/cavalier_contours cavalier_contours-master
cd C:\cavalier_contours-master
cargo build
cargo build --release
cbindgen --crate cavalier_contours_ffi --output cavc_ffi.h
```
and
`xcopy "C:\cavalier_contours-master\target\release\*cavalier_contours*.*" /y`
and
`xcopy "C:\cavalier_contours-master\target\debug\*cavalier_contours*.*" /y`

And expected that latest master will be used (checked manually, yes, i use latest)

Anyway, i took input data from my logs as intermediate result, i will check data separately, also i will check directories and dataflow more carefully tomorrow, thank you!

jbuckmccready, 2023-03-19:
Here is how I tested inputs:
Using web demo app: https://jbuckmccready.github.io/cavalier_contours_web_demo_page/#/pline_boolean
[image]

Then I use "COPY TEST CODE" button and run test under debug build in Rust.
[image]

tredecimguttatus, 2023-03-20:
Hello again!
I think i got it.
Could you please check this data:

```
{
  "name": "source_a",
  "isClosed": true,
  "Area": 11.3862,
  "vertexes": [
    [71.4473518027978684585832525, 41.0152087808049046202540922, 0.0000000000000000000000000],
    [75.0043418027978674444966600, 45.4614487808049076988936577, 1.0000000000000000000000000],
    [74.0126381972021221145041636, 46.2548112191950977489796060, 0.0000000000000000000000000],
    [70.4556481972021231285907561, 41.8085712191950946703400405, -0.2275911525975401472621229],
    [68.9692099374071290185383987, 41.0950099999999949318407744, 0.9999999999999998889776975],
    [68.9692100625928787849261425, 39.8250099999999989108800946, 0.2276528214201760380586137]
  ]
}
{
  "name": "source_b",
  "isClosed": true,
  "Area": 9.39377,
  "vertexes": [
    [62.5700009923099855768668931, 39.8250000000007773337529215, 0.0000000000000000000000000],
    [68.9692109923099962998094270, 39.8250100000007734024620731, 1.0000000000000000000000000],
    [68.9692090076900115036551142, 41.0950099999992204402587959, 0.0000000000000000000000000],
    [62.5699990076900149915672955, 41.0949999999992243715496443, 1.0000000000000000000000000]
  ]
}
```

Looks like i gave you wrong vector. sorry(

Best regards,
Andrei Rodionov

tredecimguttatus, 2023-03-20:
And another assert with this dataset:
```
{
  "name": "source_a",
  "isClosed": true,
  "Area": 1.52688,
  "vertexes": [
    [15.4607448447991053797068162, 20.4413532936729609446047107, 0.2839154127006927041954043],
    [14.5489900000002716495828281, 18.9650995449645272117322747, 0.9999999999999998889776975],
    [15.3109899999997285391373225, 18.9651004550354720379345963, -0.2839154127006926486842531],
    [15.8019351552008942007887526, 19.7600067063270401490626682, 0.9999999999999998889776975]
  ]
}
{
  "name": "source_b",
  "isClosed": true,
  "Area": 1.11524,
  "vertexes": [
    [14.5489900000254550604950055, 18.9650955958848683735595841, 0.0000000000000000000000000],
    [14.5490000000254546819178358, 18.0999955958848701698116201, 1.0000000000000000000000000],
    [15.3109999999745447496479756, 18.1000044041151326723593229, 0.0000000000000000000000000],
    [15.3109899999745451282251452, 18.9651044041151308761072869, 1.0000000000000000000000000]
  ]
}
```
Assert:
```
thread '<unnamed>' panicked at 'point does not lie on the line defined by p0 to p1 (based on distance)', C:\cavalier_contours-master\cavalier_contours\src\core\math\base_math.rs:272:5
stack backtrace:
   0:     0x7ffd5898efcf - std::backtrace_rs::backtrace::dbghelp::trace
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\..\..\backtrace\src\backtrace\dbghelp.rs:98
   1:     0x7ffd5898efcf - std::backtrace_rs::backtrace::trace_unsynchronized
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\..\..\backtrace\src\backtrace\mod.rs:66
   2:     0x7ffd5898efcf - std::sys_common::backtrace::_print_fmt
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\sys_common\backtrace.rs:66
   3:     0x7ffd5898efcf - std::sys_common::backtrace::_print::impl$0::fmt
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\sys_common\backtrace.rs:45
   4:     0x7ffd589a196a - core::fmt::write
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\core\src\fmt\mod.rs:1196
   5:     0x7ffd5898d889 - std::io::Write::write_fmt<std::sys::windows::stdio::Stderr>
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\io\mod.rs:1654
   6:     0x7ffd58990c1b - std::sys_common::backtrace::_print
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\sys_common\backtrace.rs:48
   7:     0x7ffd58990c1b - std::sys_common::backtrace::print
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\sys_common\backtrace.rs:35
   8:     0x7ffd58990c1b - std::panicking::default_hook::closure$1
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\panicking.rs:295
   9:     0x7ffd5899080e - std::panicking::default_hook
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\panicking.rs:314
  10:     0x7ffd58991211 - std::panicking::rust_panic_with_hook
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\panicking.rs:698
  11:     0x7ffd58991092 - std::panicking::begin_panic_handler::closure$0
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\panicking.rs:586
  12:     0x7ffd5898f8d7 - std::sys_common::backtrace::__rust_end_short_backtrace<std::panicking::begin_panic_handler::closure_env$0,never$>
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\sys_common\backtrace.rs:138
  13:     0x7ffd58990da9 - std::panicking::begin_panic_handler
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\std\src\panicking.rs:584
  14:     0x7ffd589a7fe5 - core::panicking::panic_fmt
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3/library\core\src\panicking.rs:142
  15:     0x7ffd588d6598 - cavalier_contours::core::math::base_math::parametric_from_point<f64>
                               at C:\cavalier_contours-master\cavalier_contours\src\core\math\base_math.rs:272
  16:     0x7ffd588d59a0 - cavalier_contours::core::math::line_circle_intersect::line_circle_intr<f64>
                               at C:\cavalier_contours-master\cavalier_contours\src\core\math\line_circle_intersect.rs:155
  17:     0x7ffd588d0065 - cavalier_contours::polyline::pline_seg_intersect::pline_seg_intr::closure$0<f64>
                               at C:\cavalier_contours-master\cavalier_contours\src\polyline\pline_seg_intersect.rs:147
  18:     0x7ffd588cd5ee - cavalier_contours::polyline::pline_seg_intersect::pline_seg_intr<f64>
                               at C:\cavalier_contours-master\cavalier_contours\src\polyline\pline_seg_intersect.rs:236
  19:     0x7ffd58921789 - cavalier_contours::polyline::internal::pline_intersects::find_intersects::closure$0<cavalier_contours::polyline::pline::Polyline<f64>,cavalier_contours::polyline::pline::Polyline<f64>,f64>
                               at C:\cavalier_contours-master\cavalier_contours\src\polyline\internal\pline_intersects.rs:358
  20:     0x7ffd589258dd - static_aabb2d_index::core::impl$8::visit<f64,tuple$<>,cavalier_contours::polyline::internal::pline_intersects::find_intersects::closure_env$0<cavalier_contours::polyline::pline [... cut ...]
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3\library\std\src\panicking.rs:456
  31:     0x7ffd588f1d0d - std::panic::catch_unwind<cavalier_contours_ffi::cavc_pline_boolean::closure_env$0,i32>
                               at /rustc/e092d0b6b43f2de967af0887873151bb1c0b18d3\library\std\src\panic.rs:137
  32:     0x7ffd589819d1 - cavalier_contours_ffi::cavc_pline_boolean
                               at C:\cavalier_contours-master\cavalier_contours_ffi\src\lib.rs:949
```

jbuckmccready, 2023-03-20:
I am able to reproduce both those cases, I looked into them briefly, they both just affect accuracy at the epsilon thresholds. The first one (assert in pline_boolean) I think indicates the code can be changed to improve accuracy in certain cases, and the other one (assert in point_from_parametric) I think is just too strict a debug assert but I need to dig into it further.

I'll work on these more and fix them when I find some time in next few days.

jbuckmccready, 2023-03-23:
For case number 2 I've tracked the issue down to `line_circle_intr` not applying the right epsilon in one of the branches, leading to loss of precision, I will commit a change to fix this right now.

For case number 1 I'm realizing the `slice_join_eps` epsilon value used now for stitching slices together (for both boolean ops and parallel offset) is maybe not required and actually leads to loss in precision (which it does for this case 2). I added the `slice_join_eps` a long time ago in the C++ code to fix issues that were arising but I'm starting to think those issues were purely due to precision/epsilon problems in the intersect code which has been fixed due to all the feed back and test cases you've provided.

The slice end points that are joined together are always at intersect points and therefore I think should be within the `pos_equal_eps` used for processing the intersects.

What's happening in case 1 is there is more than 1 candidate slice to join to when using `slice_join_eps = 1e-4` and `pos_equal_eps = 1e-5` and the code currently picks arbitrarily between them when it should pick the nearest point. I could change the code to sort and pick the nearest point but the problem is also fixed by having `slice_join_eps = pos_equal_eps`. All my test cases pass when just using the same value for both epsilons in all cases but I'm still thinking about why it would or wouldn't be required to have a larger epsilon value for the slice joining.

jbuckmccready, 2023-03-26:
OK I removed the `slice_join_eps` from the pline boolean options (breaking change, if interacted with options directly instead of using defaults), see merge commit for details.

Both cases you reported should be fixed and I added them as test cases. Thanks for the bug report. Closing this issue, please make a new issue if you find more problem cases.

closed 2023-03-26 by jbuckmccready

zdila, 2024-01-24:
Hello

I still have a problem when running my code in debug mode.

```
cavalier_contours = "0.3.0"
```

```
Polyline { vertex_data: [PlineVertex { x: 45.05580172313815, y: 21.126841631294923, bulge: 0.0 }, PlineVertex { x: 45.29628023875526, y: 21.978451356233762, bulge: 0.0 }, PlineVertex { x: 45.3785556294714, y: 22.31438601188941, bulge: 0.0 }, PlineVertex { x: 45.48890719195993, y: 24.066170448686535, bulge: 0.0 }, PlineVertex { x: 45.36683687946474, y: 26.21141480787847, bulge: 0.0 }, PlineVertex { x: 45.1854403950129, y: 27.913470684423913, bulge: 0.0 }, PlineVertex { x: 45.08583102001116, y: 28.62530354769791, bulge: 0.0 }, PlineVertex { x: 44.89833101991689, y: 29.52359223762525, bulge: 0.0 }, PlineVertex { x: 44.546768519851305, y: 30.562747757060407, bulge: 0.0 }, PlineVertex { x: 44.50770601983725, y: 31.156495331535297, bulge: 0.0 }, PlineVertex { x: 44.75770601992233, y: 31.570510944402656, bulge: 0.0 }, PlineVertex { x: 45.37489352008594, y: 31.984152120408453, bulge: 0.0 }, PlineVertex { x: 49.85145602108028, y: 34.1405290965634, bulge: 0.0 }, PlineVertex { x: 50.63246188070403, y: 34.437585093193064, bulge: 0.0 }, PlineVertex { x: 51.507706021500816, y: 34.6173131181892, bulge: 0.0 }, PlineVertex { x: 55.398331022406154, y: 35.367610650243165, bulge: 0.0 }, PlineVertex { x: 56.882706022745346, y: 35.617584708988886, bulge: 0.0 }, PlineVertex { x: 57.89833102298897, y: 35.898945306152, bulge: 0.0 }, PlineVertex { x: 59.07801852331356, y: 36.437751715665826, bulge: 0.0 }, PlineVertex { x: 59.96864352349027, y: 36.93732347214541, bulge: 0.0 }, PlineVertex { x: 60.67176852364582, y: 37.75001257927289, bulge: 0.0 }, PlineVertex { x: 61.35926852381279, y: 38.57054530057064, bulge: 0.0 }, PlineVertex { x: 62.12489352398351, y: 39.67990367306723, bulge: 0.0 }, PlineVertex { x: 62.890518524142045, y: 41.375129985092315, bulge: 0.0 }, PlineVertex { x: 63.14051852422713, y: 42.42992258604514, bulge: 0.0 }, PlineVertex { x: 63.247696258608414, y: 43.42530141100045, bulge: 0.0 }, PlineVertex { x: 63.10170016485827, y: 45.05361789236676, bulge: 0.0 }, PlineVertex { x: 63.15980563360115, y: 47.433299718476064, bulge: 0.0 }, PlineVertex { x: 63.10170016485827, y: 48.837408300659774, bulge: 0.0 }, PlineVertex { x: 62.916397430473246, y: 50.251222275592795, bulge: 0.0 }, PlineVertex { x: 62.35487399280774, y: 52.950670129473934, bulge: 0.0 }, PlineVertex { x: 62.34950289906433, y: 53.26899730323143, bulge: 0.0 }, PlineVertex { x: 62.235733367811164, y: 53.66391664410099, bulge: 0.0 }, PlineVertex { x: 62.237198211560475, y: 54.249755952534656, bulge: 0.0 }, PlineVertex { x: 62.45961032093204, y: 54.79561630515508, bulge: 0.0 }, PlineVertex { x: 62.891250946016704, y: 55.164379515744116, bulge: 0.0 }, PlineVertex { x: 63.3956454774348, y: 55.36239800628553, bulge: 0.0 }, PlineVertex { x: 64.30360446197021, y: 55.276839095450896, bulge: 0.0 }, PlineVertex { x: 65.3380282903462, y: 55.046315576115376, bulge: 0.0 }, PlineVertex { x: 66.19349704060043, y: 54.8434397351225, bulge: 0.0 }, PlineVertex { x: 66.93910250951029, y: 54.804583199059834, bulge: 0.0 }, PlineVertex { x: 68.66468844742654, y: 54.83596732479015, bulge: 0.0 }, PlineVertex { x: 70.2245029008795, y: 55.08928189054234, bulge: 0.0 }, PlineVertex { x: 71.87001071380104, y: 55.474483865584915, bulge: 0.0 }, PlineVertex { x: 73.01698337030172, y: 55.87537715595661, bulge: 0.0 }, PlineVertex { x: 73.98866305804172, y: 56.34277372506159, bulge: 0.0 }, PlineVertex { x: 75.45155368337075, y: 57.32576039830752, bulge: 0.0 }, PlineVertex { x: 76.54603610550495, y: 58.194790196311516, bulge: 0.0 }, PlineVertex { x: 77.8936923558315, y: 59.685508312162256, bulge: 0.0 }, PlineVertex { x: 78.35926852778326, y: 60.299352466161565, bulge: 0.0 }, PlineVertex { x: 79.26893649678628, y: 61.828911656715036, bulge: 0.0 }, PlineVertex { x: 79.66395602812602, y: 62.74313017067405, bulge: 0.0 }, PlineVertex { x: 80.30946384078032, y: 65.04453800751986, bulge: 0.0 }, PlineVertex { x: 81.00257907532372, y: 66.45600644854366, bulge: 0.0 }, PlineVertex { x: 81.62928805987002, y: 67.1019613313853, bulge: 0.0 }, PlineVertex { x: 83.462540013397, y: 68.77941774667417, bulge: 0.0 }, PlineVertex { x: 84.23304782606948, y: 69.68613553299747, bulge: 0.0 }, PlineVertex { x: 86.28089938906514, y: 72.26990585506496, bulge: 0.0 }, PlineVertex { x: 86.79774509227926, y: 73.1870697909057, bulge: 0.0 }, PlineVertex { x: 87.92884860823148, y: 75.32212299335848, bulge: 0.0 }, PlineVertex { x: 88.57240329589482, y: 76.73651430741478, bulge: 0.0 }, PlineVertex { x: 88.83558688965057, y: 77.80944324041889, bulge: 0.0 }, PlineVertex { x: 89.05043063972789, y: 78.96941102167264, bulge: 0.0 }, PlineVertex { x: 88.99183688973118, y: 80.26647565810416, bulge: 0.0 }, PlineVertex { x: 88.87464938965245, y: 81.09544240697952, bulge: 0.0 }, PlineVertex { x: 88.45545993643692, y: 82.684629072804, bulge: 0.0 }, PlineVertex { x: 87.91908298324014, y: 83.80609335231964, bulge: 0.0 }, PlineVertex { x: 87.4412997799426, y: 84.3813926011504, bulge: 0.0 }, PlineVertex { x: 86.24183688906326, y: 85.72736283680987, bulge: 0.0 }, PlineVertex { x: 84.66200290427633, y: 87.0243877186962, bulge: 0.0 }, PlineVertex { x: 84.10633884169336, y: 87.5705404984065, bulge: 0.0 }, PlineVertex { x: 83.64784274785514, y: 88.27246811037534, bulge: 0.0 }, PlineVertex { x: 83.44618259153378, y: 88.87278441776232, bulge: 0.0 }, PlineVertex { x: 83.30677829465553, y: 89.15034143608858, bulge: 0.0 }, PlineVertex { x: 83.267 [... cut ...]
             at /home/martin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cavalier_contours-0.3.0/src/polyline/internal/pline_offset.rs:1511:22
  12: cavalier_contours::polyline::traits::PlineSource::parallel_offset_opt
             at /home/martin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cavalier_contours-0.3.0/src/polyline/traits.rs:1387:9
  13: cavalier_contours::polyline::traits::PlineSource::parallel_offset
             at /home/martin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cavalier_contours-0.3.0/src/polyline/traits.rs:1351:9
  14: maprender::draw::draw_line_off
             at ./src/draw.rs:113:15
```

jbuckmccready, 2024-01-24:
@zdila Can you make a new issue and also try using the latest on master branch?

zdila, 2024-01-25:
@jbuckmccready I apologize, with `cavalier_contours = { git = "https://github.com/jbuckmccready/cavalier_contours.git", rev = "78ca697770ea2d97e54122bf67de303c277a662b" }` it works. Thank you.

jbuckmccready, 2024-02-22:
@zdila I made a new release, `0.4.0` with all the latest changes/fixes on master branch.