jbuckmccready/cavalier_contours #32: Offset failed, result not closed

FishOrBear, 2023-06-27:
```
{
  "isClosed": true, 
  "vertexes": [
    [0, 0, 0], 
    [0, 0, -0.001715931999999954], 
    [1.082450191001044, 50.871299999998655, 0.014190137999999974], 
    [6.351755575999277, 124.39169999999973, 0.13371957999999903], 
    [-15.339867047001007, 182.8941999999996, 0], 
    [-15.339867047000553, -20.589200000000005, 0], 
    [-33.35986704700099, -20.589200000000005, 0], 
    [-33.36040773300101, 183.4379, 0.21579646799999988], 
    [-52.47912682400147, 101.70770000000002, -0.013955579999999999], 
    [-47.65851889200076, -17.30449999999999, -0.0021751230000001134], 
    [-48.34986704700259, -128.5892000000058, -0.00033364200000000156], 
    [-45.9498670480034, -158.58919999999995, 0.08404565599999979], 
    [-55.54986704700127, -198.78920000000002, 0], 
    [-45.949867047001135, -198.7892, 0], 
    [-45.949867047001135, -207.7892, 0], 
    [-2.749867047000407, -207.7892, 0], 
    [-2.749867047000407, -198.7892, 0], 
    [9.029562464999799, -198.7892, 0.0924407319999996], 
    [19.402750884686043, -150.9626747345024, -0.0001571199999999693], 
    [20.47087665042227, -125.32843865163449, -0.00015608899999995686], 
    [22.073016405601493, -87.94516139964372, 0]
  ]
}
```

The offset algorithm in the WebCAD I implemented also has a similar error.
It is a precision problem that causes a small part of a line to be cropped.

jbuckmccready, 2023-06-28:
The first two vertexes have the same position which the algorithm doesn't automatically handle. You can run the polyline through the `remove_repeat_pos` function (https://docs.rs/cavalier_contours/0.3.0/cavalier_contours/polyline/trait.PlineSource.html#method.remove_repeat_pos) to remove all repeat position vertexes before offsetting it.

FishOrBear, 2023-06-30:
```
{
  "isClosed": true, 
  "vertexes": [
    [0, 0, -0.001715931999999954], 
    [1.082450191001044, 50.871299999998655, 0.014190137999999974], 
    [1.3777985903544732, 115.12275516020264, 0.13371957999999903], 
    [-15.339867047001007, 182.8941999999996, 0], 
    [-15.339867047000553, -20.589200000000005, 0], 
    [-33.35986704700099, -20.589200000000005, 0], 
    [-90.70415290583514, 336.0753111811248, 0.21579646799999988], 
    [-820.4762832907065, -263.75919757008427, -0.013955579999999999], 
    [-47.65851889200076, -17.30449999999999, -0.0021751230000001134], 
    [-48.34986704700259, -128.5892000000058, -0.00033364200000000156], 
    [-45.9498670480034, -158.58919999999995, 0.08404565599999979], 
    [-55.54986704700127, -198.78920000000002, 0], 
    [-45.949867047001135, -198.7892, 0], 
    [-45.949867047001135, -207.7892, 0], 
    [-2.749867047000407, -207.7892, 0], 
    [5.25083195086836, -195.1117297749882, 0], 
    [13.397706761904828, -186.0459866259352, 0.0924407319999996], 
    [7.3559734615644174, -158.18686258780286, -0.0001571199999999693], 
    [0.9559054775814957, -128.3244426226899, -0.00015608899999995686], 
    [2.643555184324782, -83.60103009313139, 0]
  ]
}
```

[image]

jbuckmccready, 2023-07-03:
Is this issue resolved? Not sure what that last polyline you posted is indicating.

FishOrBear, 2023-07-03:
The first data problem can be said to be solved, and the second is a newly discovered problem.

The second seems to be my own problem, the number of this offset needs to be set manually.

closed 2023-07-03 by FishOrBear