jbuckmccready/cavalier_contours #34: API (and C FFI) for Multi Polyline Offset

jbuckmccready, 2023-07-27:
Add C FFI wrapper function for the multi polyline offset algorithm (currently in Rust the `Shape::parallel_offset` method).

The way that the multi polyline offset is implemented requires spatial indexes for all the polylines and the bounding boxes of the polylines which will require a little more involved C FFI wrapper or a more simplistic wrapper which has to rebuild the indexes.

Current questions I'm mulling over:
- Does it make sense to have a shape type? Maybe these composite algorithms should just stand on their own as free functions? Raises question for how the data can be elegantly passed in.
- Rust API maybe needs to create traits to represent a shape type (or access to shape data) to avoid having to copy and allocate into an owned Shape type when polylines and indexes are already owned by other user structures?
- The API in Rust maybe affects how the C FFI is formed?

bdurbrow, 2024-03-05:
Any guess for when this might be done?

I'm currently working on integrating this library into a Mac application. I'd prefer to avoid duplication of effort if it's close to being ready; but if it's not then I'll just hack something together.

jbuckmccready, 2024-03-06:
> Any guess for when this might be done?
> 
> I'm currently working on integrating this library into a Mac application (FWIW, I'm using the C FFI from Objective-C). I'd prefer to avoid duplication of effort if it's close to being ready; but if it's not then I'll just hack something together.

I haven't started or done anything for the C FFI, feel free to add them and make a PR. I wont be able to get to this anytime soon.

bdurbrow, 2024-12-27:
Ugh... _finally_ got a chance finish this (came down with carpel tunnel in both hands, couldn't type - I'm a bit better now, but I still need to get the surgery done to fix it). Anyways.... 

OK, I _think_ I did this right... at least, it's working on my machine. I'm **NOT** a Rust programmer so there's probably better ways to accomplish this, but... it is working for me. Nonetheless you should probably have a look at the PR before accepting it.

jbuckmccready, 2025-01-03:
Hey, thanks for the PR! I reviewed it, link to PR for reference [here](https://github.com/jbuckmccready/cavalier_contours/pull/47).