jbuckmccready/cavalier_contours #36: corner cases result

tredecimguttatus, 2024-01-13:
Hello!

I've checked some corner cases i guess that result for 2 vectors that share border (not actually invalid, but) could be more precise.

Vectors:
```
{
  "name": "A",
  "isClosed": true,
  "Area": 142.069,
  "vertexes": [
    [7.5000000000000000000000000, 12.0000000000000000000000000, 0.0000000000000000000000000],
    [52.5000000000000000000000000, 12.0000000000000000000000000, 1.0000000000000000000000000],
    [52.5000000000000000000000000, 15.0000000000000000000000000, 0.0000000000000000000000000],
    [7.5000000000000000000000000, 15.0000000000000000000000000, 1.0000000000000000000000000]
  ]
}
{
  "name": "B",
  "isClosed": true,
  "Area": 142.069,
  "vertexes": [
    [7.5000000000000000000000000, 9.0000000000000000000000000, 0.0000000000000000000000000],
    [52.5000000000000000000000000, 9.0000000000000000000000000, 1.0000000000000000000000000],
    [52.5000000000000000000000000, 12.0000000000000000000000000, 0.0000000000000000000000000],
    [7.5000000000000000000000000, 12.0000000000000000000000000, 1.0000000000000000000000000]
  ]
}
```
Test code:
```
(
    pline_closed![(7.5, 12.0, 0.0),
(52.5, 12.0, 1.0),
(52.5, 15.0, 0.0),
(7.5, 15.0, 1.0)],
    pline_closed![(7.5, 9.0, 0.0),
(52.5, 9.0, 1.0),
(52.5, 12.0, 0.0),
(7.5, 12.0, 1.0)]
)
=>
[
    (BooleanOp::Or, &[PlineProperties::new(6, 284.137166941154, 108.84955592153875, 6.0, 9.0, 54.0, 15.0), PlineProperties::new(2, 0.0, 90.0, 7.5, 12.0, 52.5, 12.0)], &[]),
    (BooleanOp::And, &[PlineProperties::new(2, 0.0, 90.0, 7.5, 12.0, 52.5, 12.0)], &[]),
    (BooleanOp::Not, &[PlineProperties::new(4, -142.06858347057704, 99.42477796076938, 6.0, 12.0, 54.0, 15.0)], &[]),
    (BooleanOp::Xor, &[PlineProperties::new(4, -142.06858347057704, 99.42477796076938, 6.0, 12.0, 54.0, 15.0), PlineProperties::new(4, 142.06858347057704, 99.42477796076938, 6.0, 9.0, 54.0, 12.0)], &[])
]
```

I would say it is valid result, but in some cases that not what user would expect.

Best regards,
Andrei.

jbuckmccready, 2024-01-13:
Hey thanks for the report, cases where polylines only touch are well defined (as seen in this case), sometime it is what is wanted, other times it is not, it really depends on the use case. In this example how would you expect the result to be different? Not having them be joined together in the `OR` case?

tredecimguttatus, 2024-01-13:
I guess they should be connected together.

```
In my code, OR operation give this result:
{
  "name": "res[0]",
  "isClosed": true,
  "Area": 0,
  "vertexes": [
    [52.5000000000000000000000000, 9.0000000000000000000000000, 0.0000000000000000000000000],
    [7.5000000000000000000000000, 9.0000000000000000000000000, 0.0000000000000000000000000]
  ]
}
{
  "name": "res[1]",
  "isClosed": true,
  "Area": 284.137,
  "vertexes": [
    [7.5000000000000000000000000, 9.0000000000000000000000000, 1.0000000000000000000000000],
    [7.5000000000000000000000000, 6.0000000000000000000000000, 0.0000000000000000000000000],
    [52.5000000000000000000000000, 6.0000000000000000000000000, 1.0000000000000000000000000],
    [52.5000000000000000000000000, 9.0000000000000000000000000, 1.0000000000000000000000000],
    [52.5000000000000000000000000, 12.0000000000000000000000000, 0.0000000000000000000000000],
    [7.5000000000000000000000000, 12.0000000000000000000000000, 1.0000000000000000000000000]
  ]
}

```
I was talking about vector res[0].
It can be filtered out very easy, so for me it is not a problem.

tredecimguttatus, 2024-08-11:
I will close this issue, as far it was just info

closed 2024-08-11 by tredecimguttatus