jbuckmccready/cavalier_contours #38: Cut not worked

happydpc, 2024-03-17:
Here is the test code, I want to cut the rect to make a rounded corner, although it doesn't fail, but the result is not correct.
``` rust
(
    pline_closed![(233.5, 0.0, 0.0),
(222.0, 233.5, 0.0),
(-233.5, 233.5, 0.0),
(-233.5, 0.0, 0.0)],
    pline_closed![(222.0, 233.5, 0.0),
(232.934, 11.486, 0.4),
(-0.283, 233.5, 0.0)]
)
=>
[
    (BooleanOp::Or, &[PlineProperties::new(4, -107701.875, 1389.7830190582713, -233.5, 0.0, 233.5, 233.5)], &[]),
    (BooleanOp::And, &[PlineProperties::new(2, 0.0, 444.566, -0.283, 233.5, 222.0, 233.5)], &[]),
    (BooleanOp::Not, &[], &[]),
    (BooleanOp::Xor, &[], &[])
]

```
[image]
[image]

happydpc, 2024-04-06:
Updated

I adjust the bulge 0.4 to 0.3998, it's ok now.  May be there's some precisions problem when I use the real calculated f64 bulge 

[image]

The real bulge value caculated by these code, the bulge I have to truncate the precision to keep the boolean working.
```rust
   let angle = d1.angle_between(d2);
    dbg!(angle.to_degrees());         //angle.to_degrees() = 92.8195695557212
    let bulge = f64_trunc_3(bulge_from_angle(PI as f64 - angle));  //if not truncate, comes the precision bug  
    dbg!(bulge);
```

jbuckmccready, 2024-06-03:
OK the issue is `bulge = 0.4` in `Polyline 2` nearly causes it to be self intersecting (which doesn't work right for the boolean algorithm), e.g., `bulge = 0.401` will return self intersections, `bulge = 0.399` will cause boolean algorithm to work, `bulge = 0.4` will return no self intersections but also boolean algorithm fails. This is likely due to issues of inconsistency between different types of intersection (line-line and line-arc).

I'll have to look into it more to see if it can be resolved.

Running this:
```rust
    let mut intrs = Vec::new();
    pline2.visit_self_intersects(&mut |intr| intrs.push(intr));
    let intrs = pline1.find_intersects(pline2);
    println!("{:?}", intrs);
```

Get the following:
```
PlineIntersectsCollection { basic_intersects: [PlineBasicIntersect { start_index1: 1, start_index2: 1, point: Vector2 { x: -0.23185000000003697, y: 233.5 } }], overlapping_intersects: [PlineOverlappingIntersect { start_index1: 1, start_index2: 2, point1: Vector2 { x: -0.283, y: 233.5 }, point2: Vector2 { x: 222.0, y: 233.5 } }] }
```

Shows the code detects overlapping segment between these segments:
```
[233.5, 0, 0] -> [222, 233.5, 0] from Polyline 1
[-0.283, 233.5, 0] -> [222, 233.5, 0] from Polyline 2
```
as `PlineOverlappingIntersect { start_index1: 1, start_index2: 2, point1: Vector2 { x: -0.283, y: 233.5 }, point2: Vector2 { x: 222.0, y: 233.5 } }`

And a single point intersect between these segments:
```
[222, 233.5, 0] -> [-233.5, 233.5, 0] from Polyline 1
[232.934, 11.486, 0.4] -> [-0.283, 233.5, 0] from Polyline 2
```

as `PlineBasicIntersect { start_index1: 1, start_index2: 1, point: Vector2 { x: -0.23185000000003697, y: 233.5 } }`.

jbuckmccready, 2026-08-20:
Using the improvements to math in the latest release demonstrates the input is self intersecting within tolerance, boolean algorithm does not handle self intersection so I'm closing this for now.

closed 2026-08-20 by jbuckmccready