jbuckmccready/cavalier_contours #45: Splitting a polygon by a polyline

dabreegster, 2024-09-28:
I'm trying to implement something like https://postgis.net/docs/ST_Split.html. But I'm stuck, because the boolean operations require both inputs to be closed. I have one closed input, and I'm trying to perform operations against an open polyline. I might be able to hack around this by artificially closing the polyline (carefully, outside the first polygon), but I'm wondering if there's a proper way to do this. Thanks!

jbuckmccready, 2025-07-06:
Nothing in the library currently does this, but I think it would be fairly easy to add.

Logically I think the existing boolean operation already does most of what's required internally.

With one closed polyline and one open polyline as inputs it would find intersects, form slices from the intersects, discard slices from the open polyline that are outside of the closed polyline, and then finally stitch all remaining slices together.

If both inputs are open polylines then it would find intersects and just return the slices formed.

dabreegster, 2025-07-07:
Hi, thanks for the reply! Support for this was added in iOverlay in https://github.com/iShape-Rust/iOverlay/issues/7, so I've been using that. I'll close this issue, unless you wanted to support it too