jbuckmccready/cavalier_contours #48: Tracking polylines thru offset operations

bdurbrow, 2025-01-08:
I have need for tracking what output polyline came from what input polylines; and to that end have added a field to handle it:

```
pub struct Polyline<T = f64> {
    #[cfg_attr(feature = "serde", serde(rename = "vertexes"))]
    /// Contiguous sequence of vertexes.
    pub vertex_data: Vec<PlineVertex<T>>,
    /// Bool to indicate whether the polyline is closed or open.
    pub is_closed: bool,
    // Vec of user-provided u64 values. Preserved across offset calls. Note that a Polyline that was
    // composed out of multiple slices of other Polylines will have userdata values from each source
    // polyline, and as such userdata values may appear repeatedly.
    pub userdata: Vec<u64>
}
```

I _believe_ I have it working properly in the offset functions, but have not yet attempted to integrate it into the other operations (booleans, etc).

This (along with the tests you asked for) is currently up in my fork of the library: https://github.com/bdurbrow/cavalier_contours

Um - does this look reasonable to you? Questions? Comments? Rotten tomatoes? ;-)

jbuckmccready, 2025-01-15:
I can see how it gets directly at what you're wanting with minimal code changes but also seems really hacky in terms of the API, it's weird having user data values on the Polyline and then some functions operate on them in a certain way while other functions do nothing with them. E.g., if I am just using a Polyline just for other operations I wouldn't want this new field "userdata".

Trying to think if there's a more elegant solution, maybe the tracking information should be in a separate metadata structure/trait that's passed along, or maybe it's fine on the existing traits but there should be different implementors like `TrackedPolyline` and `Polyline` - both types implement the traits but `Polyline` is a no-op for userdata methods and has no "userdata" Vec field while `TrackedPolyline` is implemented how you have it. Then the `Shape` type would use the `TrackedPolyline` (or possibly is generic on the type, `Shape<P: PlineSource>` so user can decide whether they want to have tracked information or not).

In terms of just the code changes as is can avoid some unneeded restrictions on the traits and unneeded allocations:

```rust
    fn get_userdata_values(&self) -> Vec<u64>;
```
 above can be changed to:

```rust
    fn get_userdata_values(&self) -> impl Iterator<Item = u64>;
```

And for these methods:

```rust
    fn set_userdata_values(&mut self, values: &Vec<u64>);

    fn add_userdata_values(&mut self, values: &Vec<u64>);
```
above can be changed to:
```rust
    fn set_userdata_values(&mut self, values: impl Iterator<Item = u64>);

    fn add_userdata_values(&mut self, values: impl Iterator<Item = u64>);
```

bdurbrow, 2025-01-15:
Well, as I said before, I'm _not_ a Rust programmer, so educating me on improvements is _very_ welcome. 🙂

For API orthogonality I was going to extend the userdata handling to all the relevant functions of the library; I just haven't gotten there yet (so, for example, doing a boolean on two polylines with userdata on them would preserve the userdata as one would expect).

I did consider trying to track things in a separate data structure, but as far as I can tell, the polylines lose their identity going through the complex transform functions (offset, boolean, etc) - which is why I ended up at adding a field: there's nothing invariant to track them by without it.

I'm not sure that any space savings from implementing an additional subtype (i.e, TrackedPolyline) to hold the userdata Vec is worth the code complexity cost; and as all the transform functions would be instantiating TrackedPolylines for their output, at least half of the overhead would already be expended.

While I certainly appreciate small footprints, I'm having a hard time thinking of a use case where there would be enough polylines in use for the extra space taken up by an empty Vec to be a significant issue, should the user have no use for it. I'm thinking just having the API be set up such that if you don't need it, it can be ignored, should be sufficient... 🤔

jbuckmccready, 2025-01-16:
OK you convinced me, makes practical sense and caller can implement their own `SlimPolyline` with no-op functions if they really want, I await your PR. 

Also I have an idea for showing it off in example and/or README.md, we can create a shape and graph diagram to show offset relations (I think we can use `petgraph` in example code to show off tracking via indexes at each offset step, petgraph also makes it easy to generate DOT graph file to create visual graph diagram as well).

And actually I realized for these:

```rust
    fn set_userdata_values(&mut self, values: impl Iterator<Item = u64>);

    fn add_userdata_values(&mut self, values: impl Iterator<Item = u64>);
```

For maximum implementer flexibility It should be:

```rust
    fn set_userdata_values(&mut self, values: impl IntoIterator<Item = u64>);

    fn add_userdata_values(&mut self, values: impl IntoIterator<Item = u64>);
```

(Iterator implements IntoIterator and this allows lazy construction of iterator state). With this definition then if implementation is no-op no allocations/work will be done by the offset code.

bdurbrow, 2025-05-17:
Contrary to rumor, I have _not_ fallen off of the end of the earth... I’ve just ended up going down several mostly-unrelated rabbit-holes. ;-) 

It’s all working interactively in my app now (and generating proper tool paths again! Woohoo!) but there’s a bit of tidying up that I want to do before assembling a PR.

bdurbrow, 2025-05-25:
OK, I just made a pull request... I _think_ I got it right.

The source from which I built the PR is working on my machine, interactively in my app and with the unit tests.

If you find anything out of order, please let me know. 🙂

bdurbrow, 2025-05-25:
Yeah - that didn't do what I thought it was going to do.

I've deleted and re-forked my repo from your master, and re-uploaded the changes, and generated a new PR (there's probably a better way to do that, but I'm not a git expert - I know enough to use other people's code... and apparently enough to really mangle a fork. Oops).

The new PR says it should merge automatically...

jbuckmccready, 2025-05-26:
I left some comments.

jbuckmccready, 2025-07-06:
Changes merged with https://github.com/jbuckmccready/cavalier_contours/pull/63

closed 2025-07-06 by jbuckmccready