jbuckmccready/cavalier_contours #61: C FFI usage example

ChiaraTurbian, 2025-04-15:
Hi, 

Congrats on the project - great work!

Is there any example of how to use the library from plain C via FFI?
I saw the C++ one, but couldn’t find anything similar for C.

Thanks in advance,
Chiara

jbuckmccready, 2025-06-15:
I added a `README.md` to the `cavalier_contours_ffi` crate: https://github.com/jbuckmccready/cavalier_contours/blob/master/cavalier_contours_ffi/README.md

ChiaraTurbian, 2025-06-17:
Hi, thank you so much for adding the README, and thanks to both of you for your replies (I'm working on Windows though)!

In the meantime, I managed to bind the library via the C API into my C++ project, and it's working beautifully - precise and efficient. Amazing work!

I just ran into what I believe might be a bug, but I’ll open a separate issue for that.

Thanks again!
Chiara

closed 2025-07-06 by jbuckmccready