jbuckmccready/cavalier_contours #65: Determining if one polyline contains another.

bdurbrow, 2025-06-17:
I've run into the need to determine if one polyline contains another; unless I am missing something (always possible) the library does not currently contain functionality to do this directly. However, the boolean infrastructure does end up doing this as preparation for computing a boolean between two polylines... so I've started adapting it to the purpose; and it is working on my end. I believe that it's currently doing more work than is strictly necessary, however; as the boolean system generates all the intersections between two polylines; when once you find one you can quit early as you now know that the result is partial containment.

I'm a little uncertain as to the best way to proceed, in the larger context of the library's design goals:
1. Create a new version of process_for_boolean() that stops early when finding the intersections.
- Advantage: doesn't pollute the boolean infrastructure with concerns from another function; doesn't introduce a branch into the intersection search loop.
- Disadvantage: DRY violation. Larger code overall, could theoretically cause a cache miss at runtime.

2. Modify process_for_boolean() to take an extra argument to cause it to stop early when searching.
- Advantage: potentially cleaner code overall in that there's only one function that does intersection searches.
- Disadvantage: That one function now has an extra argument that all callers must pass, and must be evaluated in the inner loop.

3: Ignore the issue, as it only occurs when one polyline is colliding with another (at least for my use, that's actually an error condition and being a little slow isn't an issue).
- Advantage: less work?
- Disadvantage: Ugly. _Yuck_.

FWIW, this is what I've got so far (yes, it's mostly identical to the top of polyline_boolean():

```
/// Determine if pline1 fully contains pline2.
pub fn polyline_contains<P, R, T>(
    pline1: &P,
    pline2: &R,
    options: &PlineBooleanOptions<T>,
) -> BooleanResultInfo
where
    P: PlineSource<Num = T> + ?Sized,
    R: PlineSource<Num = T> + ?Sized,
    T: Real,
{
    if pline1.vertex_count() < 2
        || !pline1.is_closed()
        || pline2.vertex_count() < 2
        || !pline2.is_closed()
    {
        return BooleanResultInfo::InvalidInput;
    }

    let constructed_index;
    let pline1_aabb_index = if let Some(x) = options.pline1_aabb_index {
        x
    } else {
        constructed_index = pline1.create_approx_aabb_index();
        &constructed_index
    };

    let boolean_info =
        process_for_boolean(pline1, pline2, pline1_aabb_index, options.pos_equal_eps);

    // helper functions to test if point is inside pline1 and pline2
    let point_in_pline1 = |point: Vector2<T>| pline1.winding_number(point) != 0;
    let point_in_pline2 = |point: Vector2<T>| pline2.winding_number(point) != 0;

    // helper functions (assuming no intersects between pline1 and pline2)
    let is_pline1_in_pline2 = || point_in_pline2(pline1.at(0).pos());
    let is_pline2_in_pline1 = || point_in_pline1(pline2.at(0).pos());

    if boolean_info.completely_overlapping() {
        return BooleanResultInfo::Overlapping;
    }
    else if !boolean_info.any_intersects() {
        if is_pline2_in_pline1() {
            return BooleanResultInfo::Pline2InsidePline1;
        }
        else if is_pline1_in_pline2() {
            return BooleanResultInfo::Pline1InsidePline2;
        }
        else {
            return BooleanResultInfo::Disjoint;
        }
    }
    else {
        return BooleanResultInfo::Intersected;
    }
}
```

bdurbrow, 2025-06-22:
OK, I've pushed a version of it to my fork's repository. I ended up going with creating a separate function to scan for intersections and stop when it finds one.

If I'm understanding the intersection processing correctly, self-intersections in a single pline won't cause this to fail. If however, by self-intersects you mean passing in the same pline or a copy thereof, this implementation is returning PlineContainmentResult::Intersected (as it's not keeping the intersection list the way the boolean function does, there's no way to determine that it should instead return PlineContainmentResult::Overlapping).

Also... I _think_ this section can be deleted; as it shouldn't matter if an intersection could potentially be detected twice, given that it stops looking at the first one it encounters.

```
let skip_intr_at_end = |intr: Vector2<T>| -> bool {
    // skip intersect at end point of pline segment since it will be found again by the
    // segment with it as its start point (unless the polyline is open and we're looking
    // at the very end point of the polyline, then include the intersect)
    (p1v2.pos().fuzzy_eq_eps(intr, pos_equal_eps)
        && (pline1.is_closed() || i1 != open1_last_idx))
        || (p2v2.pos().fuzzy_eq_eps(intr, pos_equal_eps)
            && (pline2.is_closed() || i2 != open2_last_idx))
};
```

jbuckmccready, 2025-06-22:
For the self intersects thing I mentioned: let's say pline1 has self intersects, when the winding number is calculated for a point on pline2 it could be 0 due to pline1 wrapping around it both directions (due to it having self intersects) while still not intersecting with it. Here's an example where winding number will be zero (so contains as calculated will be false):

[image]

One could interpret this as containing, but according to winding number it is not.

jbuckmccready, 2025-06-22:
> Also... I think this section can be deleted; as it shouldn't matter if an intersection could potentially be detected twice, given that it stops looking at the first one it encounters.

Yeah, that makes sense.

bdurbrow, 2025-06-23:
Right... that's very similar to what I was thinking.

jbuckmccready, 2025-06-23:
Oops line in doc comment example didn't get committed... Fixed with latest commit: https://github.com/jbuckmccready/cavalier_contours/commit/51512bc28280c8bcc5180d73eeb8b7efc4c932dd

bdurbrow, 2025-07-02:
OK, I've implemented the required functions, they are working both in the basic unit tests and interactively in my app, and there's a pull request for them up now.

I ended up implementing separate options structs & results enums for the various functions... it's a bit of a DRY violation, but trying to cram it all onto the boolean options ended up somewhat awkward. At this point, trying to de-specialize the boolean options struct would be a breaking change for anybody using the boolean API.

jbuckmccready, 2025-07-04:
Yeah I think it's fine to define new types for this for now, trying to merge the flow for both going to require deeper changes/API design for it to make sense.

I left some comments on the PR.