jbuckmccready/cavalier_contours #66: Offset with self-intersections on multipolyline

ChiaraTurbian, 2025-06-17:
Hi,

First of all, amazing work — thanks for sharing this library!

I'm using the library via the C API in a C++ project, and I [forked ](https://github.com/ChiaraTurbian/cavalier_contours) the repo to expose the offset functionality for multipolylines in the `cavalier_contours_ffi` crate.

I believe I may have found a bug, and I was able to reproduce it in the [Web Demo](https://cavaliercontours.dev/#/multi_pline_offset) to make sure it's not related to the bindings.

I have the following multipolyline (in red), representing a rectangle with two holes. When I run an offset of `0.8`, the resulting shape (in blue) shows unhandled self-intersections.

[image]

```
[
  {
    "isClosed": true, 
    "vertexes": [
      [511.25220437557994, 328.84948025435654, 0], 
      [561.2119896118824, 328.84948025435654, 0], 
      [561.2119896118824, 363.8703101013724, 0], 
      [511.25220437557994, 363.8703101013724, 0]
    ]
  },
  {
    "isClosed": true, 
    "vertexes": [
      [540.0335350561843, 343.6169427142472, -0.2382488851276809], 
      [537.4421349268171, 345.12517844750175, -0.009889532389405053], 
      [537.3232102367999, 345.3220639672001, 0], 
      [535.3578079577983, 348.7262405716385, 0], 
      [535.32462892643, 348.7834560296831, -0.011646639385887355], 
      [535.2271073347746, 348.9562631479, -0.25910007835503845], 
      [535.2805330874, 352.1843242602999, 0], 
      [537.0000084336, 355.1625429223, 0], 
      [543.6691202685, 343.6113023833, 0], 
      [540.2257285374734, 343.6113053474554, 0], 
      [540.16153451323, 343.6120105305873, 0]
    ]
  },
  {
    "isClosed": true, 
    "vertexes": [
      [535.4816659760771, 346.2417647657877, -0.23822264248219718], 
      [535.4722003945319, 343.2448614622984, -0.009951542143624231], 
      [535.3601905012999, 343.0416179385001, 0], 
      [533.3951100416035, 339.6379987413748, 0], 
      [533.3623248097243, 339.5809609408148, -0.011710653864622287], 
      [533.2619655102294, 339.4110081813505, -0.11747560444569163], 
      [532.1675755117166, 338.3268271911126, -0.13757287100703242], 
      [530.4835288827071, 337.8423335382348, 0], 
      [530.438959126, 337.8420348466, 0], 
      [527.0000084336, 337.8420348466, 0], 
      [533.6691202683, 349.3932753857, 0], 
      [535.3908302312102, 346.4111802353502, 0], 
      [535.4223097645596, 346.3552449203639, 0]
    ]
  }
]
```

Should I preprocess the polylines in some specific way? Am I missing something, or could this be a bug?

Thank you in advance,
Chiara

jbuckmccready, 2025-06-18:
Hey thanks for reporting the issue, yeah definitely looks like a bug. From quick initial look it seems like maybe bug in stitching the polyline together after slicing at intersects related to threshold decision making at intersect points (e.g., offset = 0.799 and offset = 0.8001 don't have the problem).

ChiaraTurbian, 2025-06-18:
Hi, if it could be of any help, I'm using the default options for offsetting, but I noticed that using a finer `offset_dist_eps` (e.g. `1e-5` instead of `1e-4`) solves the issue.

I understand how tricky it is to set a general precision threshold. I also tried tweaking the parameters a bit while testing the library, but in the end your default configuration turned out to be the most robust overall.

Let me know if you find any way to handle this, thanks again!

jbuckmccready, 2025-07-06:
Hey, update on this, I plan to look into it more, I've been updating the dev/demo UI [here](https://jbuckmccready.github.io/cavalier_contours/) for being able to debug these issues better. I added editing (via json as well) so it's easy to paste in this example. You can access it with the "Edit Polylines" button.

You can see from the modes that it looks like the intersects are fine, and it creates the slices OK, but it is a bug in the slice processing not removing a slice that it should (red slice that is between the shapes in the image) before stitching them together (not 100% sure, just looking at it now).

Here's the json I pasted in to reproduce with the new UI:
```json
{
  "polylines": [
  {
    "isClosed": true, 
    "vertexes": [
      [511.25220437557994, 328.84948025435654, 0], 
      [561.2119896118824, 328.84948025435654, 0], 
      [561.2119896118824, 363.8703101013724, 0], 
      [511.25220437557994, 363.8703101013724, 0]
    ]
  },
  {
    "isClosed": true, 
    "vertexes": [
      [540.0335350561843, 343.6169427142472, -0.2382488851276809], 
      [537.4421349268171, 345.12517844750175, -0.009889532389405053], 
      [537.3232102367999, 345.3220639672001, 0], 
      [535.3578079577983, 348.7262405716385, 0], 
      [535.32462892643, 348.7834560296831, -0.011646639385887355], 
      [535.2271073347746, 348.9562631479, -0.25910007835503845], 
      [535.2805330874, 352.1843242602999, 0], 
      [537.0000084336, 355.1625429223, 0], 
      [543.6691202685, 343.6113023833, 0], 
      [540.2257285374734, 343.6113053474554, 0], 
      [540.16153451323, 343.6120105305873, 0]
    ]
  },
  {
    "isClosed": true, 
    "vertexes": [
      [535.4816659760771, 346.2417647657877, -0.23822264248219718], 
      [535.4722003945319, 343.2448614622984, -0.009951542143624231], 
      [535.3601905012999, 343.0416179385001, 0], 
      [533.3951100416035, 339.6379987413748, 0], 
      [533.3623248097243, 339.5809609408148, -0.011710653864622287], 
      [533.2619655102294, 339.4110081813505, -0.11747560444569163], 
      [532.1675755117166, 338.3268271911126, -0.13757287100703242], 
      [530.4835288827071, 337.8423335382348, 0], 
      [530.438959126, 337.8420348466, 0], 
      [527.0000084336, 337.8420348466, 0], 
      [533.6691202683, 349.3932753857, 0], 
      [535.3908302312102, 346.4111802353502, 0], 
      [535.4223097645596, 346.3552449203639, 0]
    ]
  }
]
}
```

[image]

[image]

[image]

closed 2025-07-06 by jbuckmccready